corecoding/Vitals #426: assign proper icons for temperature items

bam80, 2024-04-21:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

Right now, selected temperature items are indistinguishable and messed up if there are many:
[image]
It would be cool if we could assign proper icons for them (they also could try to be assigned automatically).

corecoding, 2024-04-23:
Can you be more specific? What is "indistinguishable and messed up"? The screenshot looks correct to me. Are you asking for custom icons? If so, what would you have them be?

corecoding, 2024-05-04:
I would be glad to review a PR with specific temperature icons. I am not a graphics designer. It actually took me a while to source these icons.

bam80, 2024-05-04:
I think existing icons from the screenshot above would suit fine.

Ticklezz, 2024-07-13:
Here is a GPU icon.
[image]

he-pennypacker, 2024-12-29:
I agree with the principle of this issue (temperatures are indistinguishable in the panel). I also appreciate that generating/sourcing additional icons (and managing their use) could be more overhead than desired.

As an alternative suggestion, how about enabling a mouseover/hover tooltip for the sensor names (e.g. processor 0, nvme 1, nvme 2 etc from the screenshot in the OP)?

bam80, 2024-12-29:
> As an alternative suggestion, how about enabling a mouseover/hover tooltip for the sensor names (e.g. processor 0, nvme 1, nvme 2 etc from the screenshot in the OP)?

While it would be better than nothing, I think it's still not enough.
The idea is you should be able to easily distinguish the temperatures items just by looking at them, before you move mouse cursor to hover etc.

binarydigit-01, 2025-04-22:
Probably grouping by component and arranging parameters next to its icon could kind of solve the issue? Like, displaying the CPU icon, and its parameters next to it: usage, frequency, temperature? Mass storage etc. accordingly?
Please, see this example, where **⚃** represents CPU and **🖸** mass storage:

**`⚃ 33%  3.2 GHz  71 °C    🖸 80%  1.7 TB  44 °C`**

Would the values/units be self-explanatory assigned to the component represented by the icon? Would that be a feasible solution? What are your thoughts?

Ariel-MX, 2026-04-25:
> Probably grouping by component and arranging parameters next to its icon could kind of solve the issue? Like, displaying the CPU icon, and its parameters next to it: usage, frequency, temperature? [...]

Or at least to be able to assign already available icons, for example, I like to add temperature readings to the top bar, CPU, GPU, RAM, NVMEs, etc.

Currently I have to remember the order I added them to the top bar to know what shown values correspond to each component, as all the temperature readings are displayed with the same thermometer icon, but in the Menu, there are icons for each type of component, so it would be great to be able to select the Processor icon as the icon for the temperature reading of CPU, and so on.

So you will end with, for example, <Processor Icon> 55ºC <GPU Icon> 30ºC <Memory Icon> 44ºC etc.