corecoding/Vitals #427: Clicking on the Temperature tab opens an invisible list

Navid-JL, 2024-04-22:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

I click on the temperate tab to see information about the temperature of different parts of my machine and it displays an invisible list. In case, you're wondering how I figured out it was invisible, adding temperature details by pressing the up/down keys and the Enter key is possible.

### Steps to reproduce the unexpected behavior.

1) Open the extension where you find different options provided by the extension such as Processor, Memory, etc.
2) Click on the Temperature.

### What distribution and version of Linux are you using?

Pop OS 22.04

### What version of Gnome are you using?

42.5

corecoding, 2024-04-23:
Can you share a screenshot? Also, if you open your terminal, what is the output of `sensors`?

Navid-JL, 2024-04-23:
[Screencast from 23-04-24 17:48:37.webm](https://github.com/corecoding/Vitals/assets/81755097/aafe8184-6416-4059-a16d-4dafa29a908f)

$ sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +46.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +46.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +47.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +45.0°C  (high = +100.0°C, crit = +100.0°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +32.0°C  

BAT0-acpi-0
Adapter: ACPI interface
in0:          12.07 V  
curr1:       530.00 mA 

nvme-pci-0200
Adapter: PCI adapter
Composite:    +23.9°C  (low  = -273.1°C, high = +78.8°C)
                       (crit = +83.8°C)
Sensor 1:     +23.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +58.0°C  
temp2:        +25.0°C  
temp3:        +34.0°C  
temp4:        +34.0°C  
temp5:        +25.0°C  
temp6:        +29.0°C  
temp7:        +35.0°C  
temp8:         +0.0°C  
temp9:         +0.0°C

corecoding, 2024-05-04:
This is not a bug with Vitals. I've heard of this report before, and had confirmation that it goes away when other extensions are disabled, I believe it was either Dash to Panel/Dock or ArcMenu. Vitals itself isn't doing anything hacky or special when it comes to creating a menu. It is using native Gnome methods/APIs. 85% of Vitals code is used to grab sensor values from /sys and /proc. If you disable other extensions, I'm positive you'll find the culprit that is causing scrollable menus to hide.

Navid-JL, 2024-05-05:
Thanks for reaching back to me. 
My apologies if it's a duplicate.
I can confirm it was **Dash to panel** that was causing this misbehaviour. 
The malfunctioning was fixed by toggling the extension in the extension manager.

I'm going to open an issue on Dash to panel's repo to address this incompatibility.

You might as well close this issue.

closed 2024-05-05 by Navid-JL