corecoding/Vitals #430: Storage not detected

perinbanm, 2024-05-11:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Storage is not measured. it is showing "Storage     No data >"

### Steps to reproduce the unexpected behavior.

click on the vital menu on the top bar and see that there are no data available for storage

### What distribution and version of Linux are you using?

Ubuntu 24.04

### What version of Gnome are you using?

46.0

capla-hunter, 2024-05-15:
Same issue in my machine too.
All other information is coming, only storage shows "No Data".
OS: Ubuntu 22.04, 
Gnome version: 42.9

corecoding, 2024-05-15:
Does this page help? https://github.com/corecoding/Vitals/wiki/Storage-reporting-%22No-Data%22

capla-hunter, 2024-05-16:
@corecoding It did worked. Thanks.

closed 2024-05-17 by corecoding