corecoding/Vitals #431: Ubuntu Studio 24.04 - fails at installation

CoolGames, 2024-05-16:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

I search for Extension Manager and then Vitals and it causes a crash message and is not installed.

### Steps to reproduce the unexpected behavior.

1. Go to search for "Extension"
2. Select and run Extension Manager
3. Turn on use User Extensions
4. Browse option to search "Vitals"
5.Select  Vitals / corecoding
6.Causes a pop up containing the Relevant log output - Version 46 not  supported

### Relevant log output

```shell
Support Log
----
Version: 0.5.0
Package Format: Unknown Package
Status: Third Party
OS: Ubuntu 24.04 LTS (Noble Numbat)
GTK: 4.14.2
Libadwaita: 1.5.0
----
(null):0 '(null)'
(null):0 '(null)'
./signal/../sysdeps/unix/sysv/linux/x86_64/libc_sigaction.c:0 '(null)'
../sysdeps/x86_64/multiarch/strcmp-avx2.S:287 '__strcmp_avx2'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
../sysdeps/nptl/libc_start_call_main.h:58 '__libc_start_call_main'
../csu/libc-start.c:360 '__libc_start_main_impl'
(null):0 '(null)'
(null):0 '(null)'
```

### What distribution and version of Linux are you using?

Ubuntu Studio 24.04

### What version of Gnome are you using?

46.0

corecoding, 2024-05-17:
Vitals on Ubuntu 24.04 is definitely supported. I personally use Ubuntu myself. That error output does not appear to be related. Vitals is written in JavaScript, not C. Try using `journalctl -f` and see if you can find mention of `Vitals`.

CoolGames, 2024-05-17:
I ran the steps as I reported and captured the journalctl -f in the included file.
Hope this helps isolate the cause.
I use barrier that shares mouse and keyboard and the installation is fresh from the USB ISO
image on May 3 2024.
The system is System 76 lemu6 from 2014 with i7 and no additional GPU.
The install had no battery condition in task bar so I wanted a tool for that.
Energy shows limited information.
Battery or AC adapter is showing age so need constant visual to predict when to plug in.
I use external HDMI and also need temp display. 

[image]

[Vitals_24.04.txt](https://github.com/corecoding/Vitals/files/15348354/Vitals_24.04.txt)

CoolGames, 2024-05-17:
After [reporting to extensions manager](https://github.com/mjakeman/extension-manager/issues/650) I again ran the steps to capture the error message here.

[image]

HeidrHH, 2025-01-29:
Hello,

I just tried to install vitals on my **TUXEDO OS 24.04.1 LTS noble** too.

I got the following error message:

```
Support Log
----
Version: 0.5.0
Package Format: Unknown Package
Status: Third Party
OS: TUXEDO OS 24.04.1 LTS
GTK: 4.14.2
Libadwaita: 1.5.0
----
(null):0 '(null)'
(null):0 '(null)'
./signal/../sysdeps/unix/sysv/linux/x86_64/libc_sigaction.c:0 '(null)'
../sysdeps/x86_64/multiarch/strcmp-evex.S:318 '__strcmp_evex'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
(null):0 '(null)'
../sysdeps/nptl/libc_start_call_main.h:58 '__libc_start_call_main'
../csu/libc-start.c:360 '__libc_start_main_impl'
(null):0 '(null)'
(null):0 '(null)'
```

Operating System: TUXEDO OS 4, based on Ubuntu 24.04.1 LTS noble
KDE Plasma Version: 6.2.5
KDE Frameworks Version: 6.10.0
Qt Version: 6.8.1
Kernel Version: 6.11.0-108013-tuxedo (64-bit)
Graphics Platform: X11

Mainboard: Asus Rog Strix X670E-A Gaming Wifi, Bios: 2704
Processors: 16 × AMD Ryzen 7 7800X3D 8-Core Processor
Memory: XPG 62.5 GiB of RAM
Graphics Processor: NVIDIA GeForce RTX 4090/PCIe/SSE2
Manufacturer: ASUS

A pity. It looks like that Vitals is what I need to monitor my pc.

corecoding, 2025-02-03:
Unfortunately Vitals is not compatible with KDE or these hybrid OSes.

closed 2025-02-03 by corecoding