corecoding/Vitals #442: Option "Monitor GPU" prevents NVIDIA from going to sleep (D3Cold)

NicolasThierion, 2024-09-03:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

The option "Monitor GPU" prevents NVIDIA GPU to enter D3Cold power save mode.

I suppose you call "nvidia-smi" to get the metrics, and the fix is not easy. 
I don't think anyone prefers having vitals on GPU rather than saving the battery, but the issue is not noticeable easily.  Maybe the option should be greyed out for laptops, or at least come with a warning.

### Steps to reproduce the unexpected behavior.

- Watch the power_state of the GPU: 
```
watch -n .2 cat /sys/class/drm/card[0,1]/device/power_state
````
- Open settings
- enable "Monitor GPU" option.
- See GPU is kept awake

### What distribution and version of Linux are you using?

Fedora

### What version of Gnome are you using?

46

egormanga, 2025-04-21:
> I suppose you call "nvidia-smi" to get the metrics, and the fix is not easy.

The fix probably _is_ easy, by simply checking that `power_state` file you mentioned first.

aaronwi, 2025-06-17:
I can confirm this is still an issue, and in fedora it causes nvidia-smi to run at 100% after standby resume, draining the battery rapidly until force killed.

Ariel-MX, 2026-04-25:
> > I suppose you call "nvidia-smi" to get the metrics, and the fix is not easy.
> 
> The fix probably _is_ easy, by simply checking that `power_state` file you mentioned first.
> 
> Upd. Although there's this thing that `nvidia-smi` is constantly running as a subprocess. I've never used an NVIDIA GPU so I can't verify it, but I see it is supposed to be a foreground monitoring tool. Is it feasible to conditionally spin up and then kill an instance at each update?

Yes, I think that would be the approach, as if nvidia-smi runs as a subprocess, it doesn't let the Nvidia card to go to a D3cold state.

So I think it is feasible to conditionally spin up and then kill an instance at each update.

I think it could be better to check:

**_/sys/bus/pci/drivers/nvidia/<ID>/power_state_**

instead:

_**/sys/class/drm/card[0,1]/device/power_state**_

If the contents of the file is D3cold, it should avoid to call the nvidia-smi program to check the GPU information.

```
ariel@graske:/sys/bus/pci/drivers/nvidia/0000:01:00.0$ pwd
/sys/bus/pci/drivers/nvidia/0000:01:00.0
ariel@graske:/sys/bus/pci/drivers/nvidia/0000:01:00.0$ cat power_state 
D3cold
ariel@graske:/sys/bus/pci/drivers/nvidia/0000:01:00.0$

```
Currently I disabled the "Monitor GPU (beta)", to prevent it to wake up the GPU when it is in D3cold state (or suspended if you ckeck:)

```
ariel@graske:/sys/bus/pci/drivers/nvidia/0000:01:00.0/power$ pwd
/sys/bus/pci/drivers/nvidia/0000:01:00.0/power
ariel@graske:/sys/bus/pci/drivers/nvidia/0000:01:00.0/power$ cat runtime_status
suspended
ariel@graske:/sys/bus/pci/drivers/nvidia/0000:01:00.0/power$

```
One interesting thing is that if you enable the **"Monitor GPU (beta)"**, it, of course, will show the GPU temp (between other useful info), focusing on the GPU temp, as en example, although it should be similar to the other GPU information, if you select to show at top bar the GPU temp (**second icon on the next image**), and disable the "Monitor GPU (beta)", the temperature will keep showing with the last known temperature, it of course will not update it, as nvidia-smi is not running, and the electable GPU option under the Temperature Menu will still showing.

[image]

If you, for example, standby the computer and wake it up, the GPU temperature icon at top bar shows only the icon and ... as you can see on the following image:

[image]

So additionally I think Vitals should behave that way when the GPU is in D3cold state, I mean, when in D3cold state, it should update the GPU data and show only ... (3 dots = no data), instead showing the last know information.

I just tested:

- Enabled  "Monitor GPU (beta)".
- Selected to show at top bar the following GPU information:

_From the Temperature Menu:_
**CPU**

_From the Graphics Menu:_
**Frequency
Memory Free**

- Disabled  "Monitor GPU (beta)".

Icons at the top bar keep showing last know data.

Standby or sleep my laptop, wake it up.

Icons at the top bar show the 3 dots (no data).

-------

Another note, I think it only applies to certain computers according to the sensors they have, in my case I noticed that even when the Nvidia GPU card is suspended (D3cold), "Monitor GPU (beta)" disabled, there is a sensor that keeps showing the GPU temperature (last row "Video"):

```
ariel@graske:/sys/bus/pci/drivers/nvidia/0000:01:00.0$ cat power_state 
D3cold
ariel@graske:/sys/bus/pci/drivers/nvidia/0000:01:00.0$

ariel@graske:~/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com$ sensors
dell_ddv-virtual-0
Adapter: Virtual device
CPU Fan:                    0 RPM
Video Fan:                  0 RPM
Chassis Motherboard Fan:    0 RPM
Memory Fan:                 0 RPM
CPU:                      +46.0°C  (low  =  +0.0°C, high =  +0.0°C)
CPU:                      +55.0°C  (low  =  +0.0°C, high =  +0.0°C)
Ambient:                  +47.0°C  (low  =  +0.0°C, high =  +0.0°C)
Ambient:                  +45.0°C  (low  =  +0.0°C, high =  +0.0°C)
Ambient:                  +48.0°C  (low  =  +0.0°C, high =  +0.0°C)
Ambient:                  +44.0°C  (low  =  +0.0°C, high =  +0.0°C)
SODIMM:                   +46.0°C  (low  =  +0.0°C, high =  +0.0°C)
Unknown:                  +29.0°C  (low  =  +0.0°C, high =  +0.0°C)
Video:                    +30.0°C  (low  =  +0.0°C, high =  +0.0°C)

```
And for I noticed, it corresponds to the temperature of the Nvidia GPU, but read for another sensor (not from the GPU information nvidia-smi provides).

So meanwhile I will disable "**Monitor GPU (beta)**" until it gets fixed to **_only run when power_state is not D3cold_**, I know I will not have all the extra GPU information, but I least I can monitor the GPU temp, even when it at suspended status.

Let me know if you need me to run more tests.

I wish I could contribute to fix the code, but I'm not a programmer.

Previously I was using the Freon gnome shell extension, and it had the same issue, I used AI to fix the related code, although it doesn't run nvidia-smi as a sub process, it was running it at the selected interval without checking if the card was in D3cold state, so used AI to fix the code.

henfiber, 2026-10-20:
I encountered this with Vitals v86, Fedora 44, kernel 7.2.11 and NVIDIA 615.78.08 on an HP OMEN MAX 16 hybrid-GPU laptop, with HDMI disconnected.

Vitals was running `nvidia-smi --query-gpu=... -l 10` in the background.

After disabling GPU monitoring in Vitals, both NVIDIA PCI functions remained `D3cold` / `runtime_status=suspended` throughout
a passive 60-second check.

In the installed `sensors.js`, `_reconfigureNvidiaSmiProcess()` starts the continuous subprocess without a PCI runtime-suspend guard.

Could NVIDIA monitoring check passive runtime state before issuing queries, avoid querying a suspended GPU, and allow idle quiet periods? The continuous subprocess needs to be stopped or paused too since ignoring its stdout would not stop polling. 

Showing “suspended” (or completely disabling the section) would be fine. 

For now, I have also disabled GPU monitoring.

MaouNour, 2026-11-16:
This happend to me 2 times today on fedora 44 on a hyprid intel + nvidia hyprid-Gpu laptop 
Disabling Gpu monitoring fixed it 
It is not a critical feature for me so I don't really mind keeping it disabled