corecoding/Vitals #445: Battery stats don't work because CMB1 is not supported

encryptededdy, 2024-10-06:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

"Monitor Battery" does not work because Battery Preferences -> Display Battery does not allow me to select my battery. My battery is "CMB1", but the menu only allows for the selection of "CMB0".

### Steps to reproduce the unexpected behavior.

Use a device with a CMB1 battery. In my case, a Fujitsu UH-X/H1. Output of `grep '' /sys/class/power_supply/*/type`;
```
/sys/class/power_supply/AC/type:Mains
/sys/class/power_supply/CMB1/type:Battery
/sys/class/power_supply/ucsi-source-psy-USBC000:001/type:USB
/sys/class/power_supply/ucsi-source-psy-USBC000:002/type:USB
```

### What distribution and version of Linux are you using?

Fedora 41 Beta Workstation

### What version of Gnome are you using?

47

RAD750, 2024-12-12:
Wouldn't it be better to enumerate what is in /sys/class/power_supply rather than hardcoding? I have a Fujitsu laptop (E5412A) that uses CMB2.

corecoding, 2025-01-20:
CMB1 and CMB2 coming in the next release. I would love to dynamically read /sys/class/power_supply - but the dynamic list in Preferences, dealing with battery entries that may disappear, etc would cause crashes that I don't have time to deal with. :)

closed 2025-01-20 by corecoding