corecoding/Vitals #450: Still seeing network sensor in toolbar when network monitoring is turned off

kozross, 2024-10-25:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

I am seeing the following in my toolbar:

[image]

However, I have the network sensor off:

[image]

### Steps to reproduce the unexpected behavior.

1. Set up Vitals
2. Note that the network sensor is shown in the toolbar
3. Disable 'monitor network'
4. Note that the sensor is not gone from the toolbar

### What distribution and version of Linux are you using?

Arch Linux

### What version of Gnome are you using?

47

corecoding, 2024-10-30:
Thank you for this report. I'm going to address this one in the next update. I am not sure if it is a regression or missed feature, but regardless should work the way you described.

skill3472, 2024-11-20:
I have the same issue, but with the "system" monitor. Waiting for a fix, thanks for the good work!

Mj65536, 2024-11-21:
Same situation in debian 12.7 (gnome 43.9 and x11)

ronzie, 2024-12-18:
Same in Ubuntu 24.10.

nebrelbug, 2025-05-12:
Same in Fedora 42

JenDot, 2025-05-13:
Fixed in G48.

corecoding, 2025-05-13:
I don't think the OS has anything to do with the bug. Vitals needs to make a pass through the list of enabled menu bar sensors and turn them off when a section is turned off. I will work on this for the next update.

LexGear, 2025-05-18:
Same issue, Zorin 17.

patrickJramos, 2025-05-27:
hey guys, i got this for a bit but after fiddling with the extension i figured out how to hide it.

first, you will need to enable "Monitor Network" again for a bit. now when you click the toolbar, a dropdown will show up. click the Network option and it will expand into a bunch of options, find the one that is selected and click it. That will hide it from the top bar.

Mj65536, 2025-06-10:
> Hey guys, i got this for a bit but after fiddling with the extension i figured out how to hide it.
> 
> First, you will need to enable "Monitor Network" again for a bit. Now when you click the toolbar, a dropdown will show up. Click the Network option and it will expand into a bunch of options, find the one that is selected and click it. That will hide it from the top bar.
> 
> You can then go back to the config and disable Network again.
> 
> It should work with any icon, just change "Network" with the one you want to hide

Superior !

corecoding, 2026-09-05:
feature has been released

closed 2026-09-05 by corecoding