corecoding/Vitals #454: support amd with redeontop

antonio24073, 2024-11-26:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

Hi, 

Please add amd support with radeontop.

Steps:

- Install radeontop
```
sudo apt install radeontop -y
```

- run the command:
```
radeontop -d --
```

It will return something like this::
```
1732580637.806896: bus 01, gpu 10.00%, ee 0.00%, vgt 0.00%, ta 6.67%, sx 7.50%, sh 0.00%, spi 10.00%, sc 9.17%, pa 3.33%, db 10.00%, cb 7.50%, vram 55.08% 563.98mb, gtt 25.40% 519.07mb, mclk inf% 0.662ghz, sclk 52.02% 0.429ghz
1732580638.807094: bus 01, gpu 10.00%, ee 0.00%, vgt 0.00%, ta 6.67%, sx 7.50%, sh 0.00%, spi 10.00%, sc 9.17%, pa 3.33%, db 10.00%, cb 7.50%, vram 55.08% 563.98mb, gtt 25.40% 519.07mb, mclk inf% 0.662ghz, sclk 52.02% 0.429ghz
1732580639.807313: bus 01, gpu 11.67%, ee 0.00%, vgt 0.83%, ta 10.00%, sx 10.00%, sh 0.00%, spi 11.67%, sc 10.83%, pa 5.00%, db 11.67%, cb 10.00%, vram 54.22% 555.25mb, gtt 25.40% 519.07mb, mclk inf% 0.668ghz, sclk 52.27% 0.431ghz
```

To get gpu usage you can do:
```
radeontop -d --| grep -Poi 'gpu \K(\d{1,3}.\d{2}%)'
```

To get vram usage, you can replace `gpu` with `vram`:
```
radeontop -d --| grep -Poi 'vram \K(\d{1,3}.\d{2}%)'
```

So you can get all the specifications from:
```
1732580639.807313: bus 01, gpu 11.67%, ee 0.00%, vgt 0.83%, ta 10.00%, sx 10.00%, sh 0.00%, spi 11.67%, sc 10.83%, pa 5.00%, db 11.67%, cb 10.00%, vram 54.22% 555.25mb, gtt 25.40% 519.07mb, mclk inf% 0.668ghz, sclk 52.27% 0.431ghz
```

To better performance, you can put the result of `radeontop -d` into a text file or cache instead run multiple parallels processes.  So run, something like `cat radeontop.txt | grep -Poi 'gpu \K(\d{1,3}.\d{2}%)'`

best regards,

antonio24073, 2024-11-28:
alternative solution for those who don't want to wait:

add a `-l 1` to the command
```
radeontop -l 1 -d --| grep -Poi 'gpu \K(\d{1,3}.\d{2}%)'
```

and put it in executor gnome extension

velias, 2024-11-29:
This issue relates to #124 where basic AMD GPU support can be provided through metrics available in sysfs (I just created a PR for that issue)

corecoding, 2024-12-02:
Thank you for your PR! Love seeing this part of Vitals being built out. Can anyone give it a test and confirm all is good?

corecoding, 2026-09-05:
feature released

closed 2026-09-05 by corecoding