corecoding/Vitals #470: Gnome 48 support

stouracp, 2025-02-11:
### Has this issue been covered in the Wiki?

- [x] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [x] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Not running on Gnome 48

error:

No property pack_start on StBoxLayout

### Steps to reproduce the unexpected behavior.

Not running on Gnome 48

### Relevant log output

```shell

```

### What distribution and version of Linux are you using?

Fedora 42

### What version of Gnome are you using?

48

corecoding, 2025-02-16:
Gnome 48 support will come. Will take a look on/after March 1st when it is in its release window.

corecoding, 2025-02-20:
A new build with support for Gnome v48 was uploaded tonight. Expect release no later than tomorrow.

closed 2025-02-20 by corecoding