corecoding/Vitals #472: settings dont worko

yabo-boye, 2025-02-28:
### Has this issue been covered in the Wiki?

- [x] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [x] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

on nobara 41 gnome 47 the gear icon for settings doesnt open anything.

### Steps to reproduce the unexpected behavior.

click on icon on panel or extensions click on gear icon 

### Relevant log output

```shell

```

### What distribution and version of Linux are you using?

nobara linux 41

### What version of Gnome are you using?

47

corecoding, 2025-03-31:
I am not seeing any logs. Can you attach some?

corecoding, 2025-08-31:
Closing due to inactivity.

closed 2025-08-31 by corecoding