corecoding/Vitals #477: Vitals stopped reporting Disk data on Fedora 42

carlosvalin94, 2025-04-15:
### Has this issue been covered in the Wiki?

- [x] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [x] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

After I upgraded from Fedora 41 to 42 (Silverblue) Vitals suddenly stopped reporting disk usage.

### Steps to reproduce the unexpected behavior.

Upgrade to Fedora 42

rpm-ostree db diff | grep sensors
  lm_sensors-libs 3.6.0-20.fc41 -> 3.6.0-22.fc42

rpm-ostree db diff | grep libgtop
  libgtop2 2.41.3-2.fc41 -> 2.41.3-3.fc42

### Relevant log output

```shell
rpm-ostree db diff | grep sensors
  lm_sensors-libs 3.6.0-20.fc41 -> 3.6.0-22.fc42

rpm-ostree db diff | grep libgtop
  libgtop2 2.41.3-2.fc41 -> 2.41.3-3.fc42
```

### What distribution and version of Linux are you using?

Fedora 42 Silverblue

### What version of Gnome are you using?

48

axeleroy, 2025-04-22:
This is probably linked to [`composefs` being enabled in Fedora 42](https://fedoraproject.org/wiki/Changes/ComposefsAtomicDesktops).

carlosvalin94, 2025-04-23:
@axeleroy  I think that could be the case, is there a way to fix this?

axeleroy, 2025-04-23:
Fixing it only requires going into Vital's settings and changing the storage path from `/` to `/sysroot`:

> ## User experience
> The main visible change will be that the root filesystem (/) is now small and full (a few MB, 100% used). The real root is mounted in `/sysroot` and most of the data is stored in `/var`. 
>
> ([Source](https://fedoraproject.org/wiki/Changes/ComposefsAtomicDesktops#User_Experience))

carlosvalin94, 2025-04-23:
It worked perfectly, thanks!!!

Last question, do you think there is a way to fix this so the extension detects the path automatically?

This way other users don't experience this issue.

axeleroy, 2025-04-23:
I honestly don't know, I have very limited knowledge on GNOME extension development and looking at the extension's code I understand that the default value is hard-coded (though I might have misunderstood).

corecoding, 2025-04-23:
I can look into seeing if I can detect composefs. Or perhaps it is as simple as seeing if the path exists, and if so, use it.

andreaxhearts, 2026-09-14:
Ubuntu 26.04, Gnome 50; I tried the path to the mount in /media and also /dev/sdX, Vitals is showing no data for me.

corecoding, 2026-09-15:
@andreaxhearts please confirm you have done this: https://github.com/corecoding/Vitals/wiki/Storage-reporting-%22No-Data%22

corecoding, 2026-09-15:
Updated Wiki with instructions specific to Fedora.

closed 2026-09-15 by corecoding